DrLex0/FFCP-GCodeSnippets #2: Number regex doesn't work in get_highest_z()

sej7278, 2019-03-05:
make_fcp_x3g wrapper script seems to use ```\d``` instead of ```[0-9]``` for digits which errors with:

```WARNING: could not find highest Z coordinate. If this is a valid G-code file, the make_fcp_x3g script needs updating.```

The fix is (PR seemed like overkill):

 ```diff
 # Path to the PWM postprocessor script in case you would be using the
 # MightyVariableFan system, my slightly crazy solution to obtain variable fan
@@ -267,7 +267,7 @@ function get_highest_z
        # Extract highest Z value in a G1 command from the last 2048 lines of a file.
        local gcode=$1
 
-       tail -n 2048 "${gcode}" | grep -oE '^G1 [^;]*Z\d*\.?\d+' | \
+       tail -n 2048 "${gcode}" | grep -oE '^G1 [^;]*Z[0-9]*\.?[0-9]+' | \
            awk '{print $2}' | cut -b 2- | sort -g | tail -n 1
 }
```

DrLex0, 2019-03-05:
Thanks for catching this. It seems this is a difference between the OS X version of grep (recognizes \d) and the one in Linux (doesn't).

Cool, GitHub has auto-closed this after I pushed my commit…

closed 2019-03-05 by DrLex0

sej7278, 2019-03-05:
yeah i've seen the osx vs linux thing before - you get the same problem with sed too.

commit e3d9b1aa (2019-03-05): Fix #2: highest Z regex fails in Linux: \d only works in OS X (BSD) grep