DrLex0/FFCP-GCodeSnippets #3: New 1.42 beta gcode

sej7278, 2019-04-23:
Just tried the updated gcode snippets on 1.41.3 (don't like 1.42beta2 at all!) and the start gcode seems quite odd.

It does the wipe and tear at the front left as usual but then proceeds to move the nozzle to the back of the bed and drop the bed and reheat the nozzle from 140 again, then do the homing and wipe again.

Assume that's not expected behaviour, although the gcode seems to show it:

```
T0; start with the right extruder. We will switch to T1 after having moved the print head to provide enough space for the nozzle offset.
M73 P0; enable show build progress
M140 S65; heat bed up to first layer temperature
M104 S140 T1; preheat nozzle to 140 degrees, this should not cause oozing
M127; disable fan
G21; set units to mm
M320; acceleration enabled for all commands that follow
G162 X Y F8400; home XY axes maximum
G161 Z F1500; roughly home Z axis minimum
G92 X118 Y72.5 Z0 E0 B0; set (rough) reference point (also set E and B to make GPX happy). This will be overridden by the M132 below but ensures correct visualisation in some programs.
G1 Z5 F1500; move the bed down again
G4 P0; Wait for command to finish
G161 Z F100; accurately home Z axis minimum
```

DrLex0, 2019-04-23:
Your diff shows changes between **right** and **left** extruders, so if you have pasted the rightmost part into your ‘Right’ printer profile, things won't work as expected because it's the start code for the left extruder.
Also if the code does the same thing twice, maybe you accidentally pasted the new code after the old one, or somewhere in the middle, instead of replacing it?

I have tested this code with both 1.41.3 and 1.42 without problems…

sej7278, 2019-04-23:
i've no idea what i've done lol, the old right extruder gcode seems to work fine when using the left extruder on 141 although i did notice that the wipe happens on the front right (where my bed bracket is) when using the right extruder which didn't seem right.

i've rebuilt my slic3rpe profile now and re-imported the profiles and gcode carefully, will try a test print in a minute.

i did notice that ctrl-a doesn't seem to work in textareas in 141 so i may have thought i deleted the existing gcode but in fact pasted the new gcode on the end - would explain why it seems to do two heat/wipe/level routines!

are the config bundles always in sync with the gcode, as re-importing the configbundle is a lot easier than copy'n'pasting the gcode for every extruder combo.

DrLex0, 2019-04-23:
The config bundles are not always in sync. Especially now that 1.42 is still in beta I'd rather postpone updating the bundle until the final 1.42 has been released.
If you're sticking to 1.41 anyway, then there is not much use in updating the G-code, as the only changes compared to what is in the bundle, are inside the comment section.

sej7278, 2019-04-23:
Sorted it - i think i must have double-pasted or something, 1.41.3 is working fine now, sorry.

i'll move to 1.42 when it goes final and/or they sort out the icon mess on linux - at the moment most of the icons are missing or corrupt-looking, probably as the wxwidgets they're linking against is using gtk2. that said i still don't like the split/rotate in 1.42b2, it was a single click and much more intuitive in 1.41

closed 2019-04-23 by sej7278