DrLex0/FFCP-GCodeSnippets #4: Ooze/wipe always on left of bed

sej7278, 2019-05-27:
I use your brackets for my glass bed and have the front one on the right hand side.

When using your gcode with ABS (right extruder) the ooze/wipe/loop doesn't work so well as I guess it interferes with the bracket.

How do I change the gcode to always use the left side for that (like PLA)? I can't see much difference between the left/right extruder gcode for the ooze/wipe, other than hyphens (e.g. X70 vs X-70):

```diff
149c145
< G1 X70 Y-74 F4000; chop off any ooze on the front of the bed
---
> G1 X-70 Y-74 F4000; chop off any ooze on the front of the bed
151c147
< G1 X-121 E24 F2000; extrude a line of filament across the front edge of the bed (3mm from the front edge)
---
> G1 X121 E24 F2000; extrude a line of filament across the front edge of the bed
154,156c150,152
< G1 X-108 Y-74 F4000; cross the extruded line to close the loop
< G1 X-100 F4000; wipe across the line (X direction)
< G1 X-90 Y-79 F6000; Move back for an additional wipe (Y direction)
---
> G1 X108 Y-74 F4000; cross the extruded line to close the loop
> G1 X100 F4000; wipe across the line (X direction)
> G1 X90 Y-79 F6000; Move back for an additional wipe (Y direction)
160c156
```

sej7278, 2019-05-27:
Would this change do it - adds a "move to front right" before the ooze chop then basically matches the left extruder gcode (e.g. X-108 instead of X108):

```diff
@@ -142,14 +142,15 @@
 M116; wait for everything to reach target temperature
 M17 A; re-enable right extruder stepper
 G1 Z0 F1000
-G1 X-70 Y-74 F4000; chop off any ooze on the front of the bed
+G1 X70 Y-83 F8400; move to front right to match left extruder gcode
+G1 X70 Y-74 F4000; chop off any ooze on the front of the bed
 G1 Z[first_layer_height] F1500; move to first layer height
-G1 X121 E24 F2000; extrude a line of filament across the front edge of the bed
+G1 X-121 E24 F2000; extrude a line of filament across the front edge of the bed
 ; Note how we extrude a little beyond the bed, this produces a tiny loop that makes it easier to remove the extruded strip.
 G1 Y-71 F2000
-G1 X108 Y-74 F4000; cross the extruded line to close the loop
-G1 X100 F4000; wipe across the line (X direction)
-G1 X90 Y-79 F6000; Move back for an additional wipe (Y direction)
+G1 X-108 Y-74 F4000; cross the extruded line to close the loop
+G1 X-100 F4000; wipe across the line (X direction)
+G1 X-90 Y-79 F6000; Move back for an additional wipe (Y direction)
 ;G92 E-0.6; This no longer works with relative E. The purpose was to compensate for the inexplicable but consistent under-extrusion that occurs at the start of the skirt. This compensation must now be done in a post-processing script.
 G1 F8400; in case Slic3r would not override this, ensure fast travel to first print move
 M73 P1 ;@body (notify GPX body has started)
```

sej7278, 2019-05-27:
edit: forget it, just realised how stupid this is - the right extruder can't get that far past the left of the bed!

closed 2019-05-27 by sej7278

DrLex0, 2019-05-27:
Yeah, that's the reason why the wipe directions are opposite for the two extruders :)
I also have one bracket on the front right, and to help avoid any issues for those (rare) cases where I print with the right extruder, I have stuck a small piece of painter's tape on top of the bracket where the chop/wipe is performed.