DrLex0/FFCP-GCodeSnippets #5: Wrong temp

zwoop, 2019-09-01:
Was hoping you might have an answer, but when using your presets, and trying to print some PLA at 205 degrees, it seems to send it to the FFCP with a target of 140 degrees. If I monitor the FFCP via Octoprint using a normal print (.x3g) it reports the correct temperature. So, something is going wrong here, even though the Filament settings shows a target of 205 in Slic3r.

Any ideas ?

DrLex0, 2019-09-01:
The extruder always starts heating at 140°C until the correct bed temperature is reached, then it will continue heating to the actual extrusion temperature. This is to avoid cooking the filament while it sits idle during the heating of the bed, which usually takes longer than heating the extruders.
If the extruder temperature does not jump to 205°C after the bed has reached 60°C (or whatever you have configured), then there is something wrong.

zwoop, 2019-09-01:
Aha! Yeh that was it. I got it working with Octoprint and Slic3r without problem, very straight forward to just import the .ini. I made one change however, instead of doing the Postprocessing in the .gcode, I installed and enabled the GPX plugin in Octoprint. And it works very well, no problems at all!

closed 2019-09-01 by zwoop