DrLex0/FFCP-GCodeSnippets #7: X-axis offset incorrect when printing with left extruder

craftyguy, 2019-09-29:
**Describe the bug**

When printing with the 'FFCP single material L' printer profile, the print is shifted to the right along the X-axis and is not centered on the build plate.

**Environment**

* Your OS and the version of PrusaSlicer the issue is encountered with.

Arch Linux, with PrusaSlicer v2.1.0

* The version of your GPX binary (try `gpx -?`)

gpx 2.5.3 (from git master branch)

* The version of your Perl runtime (try `perl --version`, obviously not needed if you are not using the Perl-based post-processing scripts).

v5.3.0

**To Reproduce**

Please attach the project file (3MF) that demonstrates the problem and if possible, a ZIP with both the final .gcode and .x3g files.
[jerker_cable_clip.zip](https://github.com/DrLex0/FFCP-GCodeSnippets/files/3667983/jerker_cable_clip.zip)

* If specific steps are required to reproduce the behavior, list them here.

I opened the attached stl in Prusa Slicer, made no changes to the position or rotation, sliced, exported gcode (including conversion to x3g with your script) and printed. The print was offset from the center of the bed. For a small print like the included example, this is not a problem, but for larger prints this is a major problem.

DrLex0, 2019-09-29:
This cannot be a problem with these settings. There is nothing particular about your files that would cause this behavior. Just to confirm this, I have printed your x3g file on my printer (with no filament loaded) and it indeed prints in the center as I expected.

The problem must be with your printer itself. Check the ‘Toolhead offsets’ in the Utilities menu of the printer's LCD menu. The X offset must be approximately 34 mm (only if you're dualstruding it is important to accurately calibrate this). Your home offsets might also be wrong, but they should be OK if the R profile is well-centered.

Have you tried using ReplicatorG recently by any chance? If you used the wrong version, it can mess up certain settings like the toolhead offsets. You must use one of [these builds](https://www.thingiverse.com/thing:32084/files).

craftyguy, 2019-09-29:
> This cannot be a problem with these settings. There is nothing particular about your files that would cause this behavior. Just to confirm this, I have printed your x3g file on my printer (with no filament loaded) and it indeed prints in the center as I expected.

Oh, I see. Sorry for filing an issue then, I wasn't sure what the problem might have been.

> The problem must be with your printer itself. Check the ‘Toolhead offsets’ in the Utilities menu of the printer's LCD menu. The X offset must be approximately 34 mm (only if you're dualstruding it is important to accurately calibrate this). 

After resetting to factory settings, the X offset shows up as 3100-something mm, which seems odd.

> Have you tried using ReplicatorG recently by any chance? 

I have not. I'm using (what I believe) is the version of sailfish that came with this thing ('Sailfish Creator Pro v7.8').

closed 2019-09-29 by craftyguy

DrLex0, 2019-09-29:
Yeah, that 3100 explains it: the printer will try to move the carriage 31 meters to the right when performing the tool change. Of course that's impossible and the endstop switch will prevent it, but by that time the carriage will have moved way beyond 34 mm. That's why the whole print shifts to the right.
You must reset this X toolhead offset to 34 mm. Most likely you'll need to do this through RepG unless you want to be holding down the ‘down’ button for several hours (days?) until it has counted down from 3100 to 34.

If you want to do accurate dual extrusions, you should further tweak this offset following [these instructions](https://www.dr-lex.be/info-stuff/print3d-dualstrusion.html#hints). If you're never going to do a dual extrusion, just setting the offset to 34.0 mm is good enough.

craftyguy, 2019-09-29:
>  Most likely you'll need to do this through RepG unless you want to be holding down the ‘down’ button for several hours (days?) until it has counted down from 3100 to 34.

Or a c-clamp (only took about an hour!) :)

Thanks for the info, I can see why this happened in the first place. I've set the X offset to 34mm (I don't plan on ever using dual extrusion..), and going to test it out now.

Thanks for the help, and for the wealth of knowledge you've made available for folks like myself who don't yet have any idea what they are doing. I appreciate it!