DrLex0/FFCP-GCodeSnippets #8: Incremental printing offset after updating make_fcp_x3g

CutAs, 2020-05-11:
System: MacOS Catalina

Hi, after upgrading my Mac to Catalina, the files in /bin got automatically moved or deleted. Therefore I re-downloaded the latest version of the make_fcp_x3g script. I didn't upgrade the rest as other scripts seem to not have been modified since my installation (my gCode configs are around 8-9 months old)

The conversion script runs fine, but after laying the first 2-3 layers correctly, FCP starts offsetting by around 2-3mm at each new layer, each time on -Y axis, moving towards the front of the printer, until the head starts bashing against the front panel.

I wonder if there might be a bug in the updated script, or did I miss something?

DrLex0, 2020-05-11:
This kind of problem looks rather as if one of your stepper cables has developed a bad connection. Do you still have an .x3g file that you successfully printed before? If so, try printing that again.

If the problem does occur only with newly generated files, please attach both a .gcode file and the resulting .x3g file so I can take a look at it.

CutAs, 2020-05-14:
Hi, thanks for the tip! 

I could not find the old x3g unfortunately, but I tried to print the same model through FlashPrint and the same effect does not seem to reproduce itself.

DrLex0, 2020-05-14:
And if you print the exact same file again that originally exhibited the problem, does it fail in the same way? Of course you could print ‘dry’ with no filament loaded to avoid wasting it, if it is obvious enough when the extruder keeps moving forward…

CutAs, 2020-05-22:
Appears that it doesn't. Maybe indeed it was a cable connector issue, if it reproduces again I will let you know! 

Thanks.

DrLex0, 2020-10-01:
Closing this, as it seems to have been caused by another issue. I recently had to replace my X axis cable because it also started skipping steps…

closed 2020-10-01 by DrLex0