DrLex0/FFCP-GCodeSnippets #9: Tool 0 temp being set while using Single Tool 1 Single printer profiles.

Julzilla, 2020-09-29:
Hey Dr Lex!

First off, big fan of your work and dedication to this project, absolutely amazing stuff!

I'm having an issues with both single tool 0 profiles (FFCP Single Material Left and Left Flex).

After slicing in PrusaSlicer, post processing in Perl and sending the GCode to OctoPrint, the file starts printing with no issues until it puts the first layer down and then Tool 0 sets itself to the Filament 'Other Layers' temp. Every time I print I need to turn Tool 0 heat to off.

I have looked through the GCode and Perl script but am unable to figure out why this is happening. 

Attached are my print settings, GCode and project file.The perl script remains unchanged.

OS and the version of PrusaSlicer:
Windows 10 (build 20221) & PrusaSlicer 2.2.0

The version of your GPX binary:
2.6.5

The version of your Perl runtime:
Strawberry Perl (64-bit) 5.32.0.1-64bit

[Profile Files and GCode.zip](https://github.com/DrLex0/FFCP-GCodeSnippets/files/5295934/Profile.Files.and.GCode.zip)

- Import STL into PrusaSlicer
- Slice
- Send to Octoprint
- Print
- Layer 1 complete - Extruder T0 sets to 'Other Layers' heat of the selected filament profile.

DrLex0, 2020-09-29:
You have set your post-processing script path to:
```
"C:\Strawberry\perl\bin\perl.exe" "C:\post scripts\retraction-improver.pl";
```
That doesn't match the instructions at all and it won't do anything. Even if putting those two paths together does run the given Perl script, it will still do nothing because that particular script dumps its result on standard output, it doesn't write any files by itself. And moreover, that is the least important of the scripts. The `make_fcp_x3g` script, or its simpler BAT alternative, are far more essential because they perform the necessary workaround to avoid [the PrusaSlicer bug](https://github.com/prusa3d/PrusaSlicer/issues/2210) that causes this heating up of the wrong tool.

All the instructions are in the README. I have made some edits in an attempt to make them less confusing, so you should go over them again and ask if something isn't clear.

closed 2020-09-29 by DrLex0

Julzilla, 2020-09-30:
I will give them another look over, all this stuff is new to me. Thanks again!

Julzilla, 2020-09-30:
All working now. Thanks for helping this noob RTFM 👍