DrLex0/FFCP-GCodeSnippets #11: Issue with converting g-code file to x3g file using make_fcp_x3g script

Kyderio, 2020-11-30:
GPX version 2.6.7, OS: Ubuntu 16.04, PrusaSlicer version 2.2.0

Hi DrLex0, 

I'm trying to use the script to automatically convert the g-code file to x3g file. I have turned debug setting on and it says I did not give it permission to be executable, but I think I did that..

Can you help me double check if I am missing anything? 

[gcode2x3g.zip](https://github.com/DrLex0/FFCP-GCodeSnippets/files/5614482/gcode2x3g.zip)

DrLex0, 2020-11-30:
Your third screenshot clearly shows that the gpx file is not executable.
Either try in the Linux console: `sudo chmod a+x /usr/local/bin/gpx`, or better, delete that file and install the standard Ubuntu package instead with `sudo apt install gpx` (this will place it in `/usr/bin/gpx` so adjust the script accordingly).

Kyderio, 2020-12-02:
Hi DrLex0,

Thanks for the fast response! However I have no idea why giving it permission as of the first method doesn't work..So I tried the second method..However the second method installs the gpx version 2.0-alpha, which is what you don't recommend using?

The attached photo is the copy of what I have input in the make_fcp_x3g script for the first method, and I have installed the latest 2.6.8 for gpx.
[image]
[image]

DrLex0, 2020-12-02:
OK, it seems the gpx version in Ubuntu 16.04 is indeed the bad one, so don't use that (you could remove it with `apt-get purge gpx`).

If:
* the gpx 2.6.8 binary is at `/usr/local/bin/gpx`,
* you executed `sudo chmod a+x /usr/local/bin/gpx` and it did not fail in any way,
* you configured `GPX=/usr/local/bin/gpx` in the script,

then it should work.

Kyderio, 2020-12-03:
Hi DrLex0,

I realised that in the script, GPX=/usr/local/bin/gpx will fail the sanity check, while the GPX= /usr/local/bin/gpx will not fail the sanity check. (Note the space between the = sign and / sign.)

Either way, after I export G-code, am I supposed to see 2 g-code files and 1 sanity check file only? I thought I am supposed to be seeing a x3g file...In the sanity check file, it is reading 'All checks seem OK!' 

[gcode2x3g.zip](https://github.com/DrLex0/FFCP-GCodeSnippets/files/5633178/gcode2x3g.zip)

DrLex0, 2020-12-03:
Adding a space turns that line into two separate commands:
```
GPX=
/usr/local/bin/gpx
```
The end result is an empty GPX variable, and gpx being invoked without arguments.
If the GPX variable is empty or undefined, the script skips the x3g conversion step altogether, which is why you aren't seeing any x3g file.

Can your run this script from within the Ubuntu shell: [test.tar.gz](https://github.com/DrLex0/FFCP-GCodeSnippets/files/5639045/test.tar.gz)

It's wrapped inside a tarball because GitHub doesn't allow attaching just anything. An advantage is that it will (should) be executable immediately after unzipping. Use `tar -xzf test.tar.gz` to decompress it, then run it with `./test.sh` and copy the output.
I am really curious what this will show because this whole problem makes no sense at all. It will probably be something really simple, as is always the case with these kind of problems.

Kyderio, 2020-12-04:
Hi DrLex0,

This is the output from the test.sh file.
[image]

It truly makes no sense to me too...

Kyderio, 2020-12-04:
Hi DrLex0,

I have finally found out why the script is not working...
Originally, I have installed the Prusaslicer through the Ubuntu Software which is also version 2.2..

Now, I've uninstalled Prusaslicer through Ubuntu Software and download it from https://www.prusa3d.com/prusaslicer/#intro.

The script is working now...

Thank you so much for your help!

Best Regards,
Kelvin

closed 2020-12-04 by Kyderio