DrLex0/FFCP-GCodeSnippets #18: Base layer versus first layer height

scotte, 2022-01-17:
Just a heads up - I'm not sure if this will apply to newer versions of Slic3r or PrusaSlicer, but with the SuperSlicer fork, the `First layer height` cannot be larger than the `Base layer height` in `Slicing` configuration under `Print Settings`. Here's the details: https://github.com/supermerill/SuperSlicer/issues/2060

DrLex0, 2022-01-19:
There is no such thing as base layer height in the current PrusaSlicer, hence there is no value for base layer height in the profiles published here, hence I there is nothing to fix.

Quoting the README: “These configs and G-code are made specifically for PrusaSlicer. They might work in the original Slic3r from which PrusaSlicer was forked, but I give no guarantees, and I give no support if you use this with anything else than PrusaSlicer.”

I don't even know what base layer height means, and why a first layer height of 0.2mm, 0.25mm, or 0.27mm would be problematic.

scotte, 2022-01-20:
Sorry if the term got muddled - you are correct, it's called `Layer height` in Prusa Slicer, `layer_height` in the configurations: https://github.com/DrLex0/FFCP-GCodeSnippets/blob/master/ConfigBundles/Slic3r_config_bundle.ini#L66

To be clear, I'm not asking for support and already solved this myself. I am also well aware that your configurations are made for Prusa Slicer, not other forks. As I do not know if this change came in the Super Slicer fork or from upstream (either Prusa Slicer or Slic3r), I was simply trying to be helpful.

DrLex0, 2022-01-20:
If I understand it right, this new slicer prohibits using a thicker first layer? A thicker first layer is often essential to get prints to start reliably. Good luck trying to deposit a 0.5 mm first layer on a print bed that is not perfectly flat and not perfectly levelled…
If you cannot configure a first layer thicker than the other layers (regardless of the minimum layer height parameter value), then that's an outright bug in that slicer and you should file a new issue there, because it does not seem that supermerill/SuperSlicer#2060 has anything to do with it.

The actual problem mentioned in #2060, of the minimum layer height not being imposed on the actual layer height, seems to be a PrusaSlicer issue and should be filed there.

closed 2022-02-14 by scotte