DrLex0/FFCP-GCodeSnippets #22: Add STL Model of the build plate [enhancement]

lthown, 2022-09-27:
There's a model of the FFCP build plate on https://www.thingiverse.com/thing:3772642

This file can be loaded into the PrusaSlicer using the "bed shape" setting

[image]

The results are quite nice, though the file appears to be too far back by a few mm

[image]

Still a nice way to visualize the print within the printer.

DrLex0, 2022-09-27:
It's not entirely useless, because it adds some visual reference points. and this avoids the risk of placing things upside down, which I sometimes end up doing. I'll see if I can tweak the model to be 100% correct and maybe also add a texture.

lthown, 2022-09-27:
cool beans

DrLex0, 2022-10-09:
Added, but as you'll see, because PS relies on absolute paths to these files, it is not possible to load the model (and texture) fully automatically while importing the ini.
I have added functionality to the config helper page that can avoid having to manually configure the files for each of the printer profiles, although in your case this is probably a better option than going through the whole process again.

This would all be simpler (at least for users) if I could distribute these configs as a vendor preset in PrusaSlicer, but this is blocked by their unwillingness to solve the issue with the tool number.

closed 2022-10-09 by DrLex0

lthown, 2022-10-09:
That's frustrating but what you put together looks fantastic