DrLex0/FFCP-GCodeSnippets #24: No X3G file exporting?

cat80k, 2023-02-20:
Hi, so I just got a FF Creator Pro with dual extruders, and I'm trying to use PrusaSlicer with it for its painting functionality.
I am fairly certain that I followed the instructions exactly, but upon exporting a GCode, there is no X3G file being created alongside it. The debug file and the momentary popup after exporting both say everything is functioning, but I am not getting an X3G file out of it.

I'm sure it's because I am not as hardcore code savvy as most 3D printer enthusiasts are, so I'm sorry ahead of time if it's something stupid.

Windows 10
PrusaSlicer 2.5.0
GPX 2.6.8
Strawberry 5.32.1.1

Everything is freshly installed, literally. I just built this PC 2 days ago for use with 3D stuff.

[bunnytest.zip](https://github.com/DrLex0/FFCP-GCodeSnippets/files/10788138/bunnytest.zip)

DrLex0, 2023-02-21:
Try updating to the current `make_fcp_x3g.pl` script and run it with `-vc` argument in a command prompt as explained in the troubleshooting section of the README. This will now output more information.

cat80k, 2023-02-21:
> Config read from 'C:\Users\atemu\Documents\3D Printing Files\FFCP-GCodeSnippets-master\make_fcp_x3g.txt'
> GPX will not be used because GPX path is not configured
> DUALSTRUDE_SCRIPT will not be used because not configured
> PWM_SCRIPT will not be used because not configured
> RETRACT_SCRIPT will not be used because not configured
> All checks seem OK!

I may be too dumb for this, I'm sorry. I don't know how to configure the GPX path, but I understand now that that is what's needed to convert the file.

DrLex0, 2023-02-21:
You have gotten pretty far for someone who claims not to understand things. The only thing missing is the correct line in the `make_fcp_x3g.txt` file specifying the path to the GPX executable.

You can obtain this path by opening the folder where you installed GPX, holding down the Shift key, and then right-clicking on *gpx.exe*. This will reveal a normally hidden “Copy as path” menu option, select this and the path to GPX will be in your clipboard.

Then open the `make_fcp_x3g.txt` file in a text editor, and paste the path after the line starting with “`GPX =`”. Remove the `#` character at the start of this line if there is one. In other words, the line should look something like this:
```
GPX = "C:\Program Files\gpx-2.6.8-win64\gpx.exe"
```
but the actual path in your case may be different, depending on where you installed GPX. Then save the txt file and try again.

cat80k, 2023-02-21:
AAAA it's working! Thank you so much for helping me. I'll be sure to donate to you on your website, too.

`You have gotten pretty far for someone who claims not to understand things.`
My only saving grace is being able to mostly follow instructions. I've done similar UI modding to MMOs before using other people's mods and instructions, but sometimes I hit a snag and need help. OTL

closed 2023-02-21 by DrLex0